IMRCLab/crazyswarm2 #438: Simulation not working

fbenti, 2024-02-23:
I'm following the instructions for launching the web-based simulator, the only difference is that I don't have a joystick connected.

I can correctly visualize the drone, however when i run `ros2 run crazyflie_examples hello_world --ros-args -p use_sim_time:=True
`, nothing happens. I can see that the command is correctly received. 
[image]

fbenti, 2024-02-23:
Is it required to be connected to the crazyflie231? I mean with the Crazyradio

knmcguire, 2024-02-23:
Hi!That is not a simulator, but just a GUI for visualization like RVIZ.

What was the commands you used to startup the crazyflie server? That's with a real crazyflie then I presume? And are you getting any errors in that server?

fbenti, 2024-02-23:
I first run `ros2 launch crazyflie launch.py backend:=sim` and everything seems fine.
[image]

Then in a second terminal
`ros2 run crazyflie_examples hello_world --ros-args -p use_sim_time:=True`.

But nothing happens. Isn't the GUI supposed to show takeoff and land in this case?

knmcguire, 2024-02-23:
Ah yes oke then you are running the simulation backend indeed. You don't need to be connected to a real crazyflie for that.

Indeed it seems that something is broken. If you run `ros2 launch crazyflie launch.py backend:=sim rviz:=True`, then you do see the crazyflie reacting on the hello world script in RVIZ but not in the web-based gui. 

So this seems to be an actual bug, good catch! I'll change the title though since it is not actually the simulator is broken.

fbenti, 2024-02-23:
Yes RVIZ respond to that.

whoenig, 2024-02-23:
Adding https://github.com/IMRCLab/crazyswarm2/blob/261f3e8ef688a21a378e7febd68e95d4ec95be73/crazyflie/launch/launch.py#L136-L138 to gui.py fixes the problem, but has issues when running in non-sim mode. At the end we need logic that adds these lines conditioned on the launch argument. This is also an issue with rviz, by the way (the "elapsed time" is wrong in non-sim mode).

PR #439 (2024-02-27): dynamically update use_sim_time, depending on backend
Fixes #438
commit 4bbe8ce9 (2024-02-27): dynamically update use_sim_time, depending on backend

Fixes #438
closed 2024-02-27 by knmcguire