IMRCLab/crazyswarm2 #442: Low frequency of custom topic (firmware logging) after enabling non-crazyflie objects in crazyflies.yaml

xiaoshan-lin, 2024-02-26:
**Descriptions**: We have a custom topic to log two parameters: pm.state and pm.vbat. The custom topic is published at a desired rate (10Hz) when running multiple crazyflies. However, if we add non-crazyflie objects in the yaml file, the topic will **significantly slow down (< 0.5 hz)**, and sometimes even **stop publishing**. We use crazyradio PA, and the firmware of the crazyflie is 23.11. The [crazyflies.yaml](https://github.com/IMRCLab/crazyswarm2/files/14409161/crazyflies.yaml.txt) is attached here. 

**Questions:**
1. We add tubtlebots as non-crazyflie objects in the yaml file as below, but not sure if we configure them correctly. Does the id of the non-crazyflie object matter? We put id 254 and 255 (see [documentation](https://imrclab.github.io/crazyswarm2/howto.html#tracking-non-robotic-objects)).
 ```
     robots:
          cf03:
             enabled: true
             uri: radio://0/90/2M/E7E7E7E703
             initial_position: [0.21, 1.28, 0.41]
             type: cf21_merl_mocap_deck  # see robot_types
          cf04:
             ... 
          turtlebot1:
             enabled: true
             initial_position: [- 0.26, - 1.42, - 0.38]
             type: turtlebot_1      # see robot_types
             id: 254               
          turtlebot2:
             enabled: true
             initial_position: [-0.91, -0.22, -0.38]
             type: turtlebot_2      # see robot_types
             id: 255       

     robot_types:
          cf21_merl_mocap_deck:
            ...   
          turtlebot_1:
             connection: none
             motion_capture:
             enabled: true
             marker: turtlebot_marker_1
             dynamics: default
          turtlebot_2:
             connection: none
             motion_capture:
             enabled: true
             marker: turtlebot_marker_2
             dynamics: default
   ```
2. Is the custom topic expected to have low frequency if enabling non-crazyflie objects, or am I doing anything wrong here?

Thank you for your help and support!

whoenig, 2024-02-26:
Does the Crazyflie firmware need to know the position of turtlebot_1 and turtlebot_2?

The only difference is that this information will be also send via broadcast to all Crazyflies. This might require an additional packet, which can slow down unicasts. 

Perhaps you can post the output of one of the /cf<id>/connection_statistics and/or of /cf<id>/status. In general the options are to i) manually configure motion_capture_tracking if you don't need the additional information on the firmware side, or ii) reduce the motion capture frequency.

xiaoshan-lin, 2024-02-26:
Thank you for responding so quickly.

The crazyswarm firmware doesn't need to know the position of turtlebot_1 and turtlebot2. 

This totally makes sense. I ran a few more tests and found there is no slowdown when I enable just 1 crazyflie and 1 turtlebot. If I increase the number of robots, the frequency will drop.

Are /connection_statistics and /status ros topics? I couldn't find them after I launch the crazyswarm.
```
>> ros2 topic list
/all/cmd_full_state
/cf04/battery_status
/cf04/cmd_full_state
/cf04/cmd_position
/cf04/cmd_vel_legacy
/cf04/pose
/cf06/battery_status
/cf06/cmd_full_state
/cf06/cmd_position
/cf06/cmd_vel_legacy
/cf06/pose
/cf6/cmd_full_state
/cf6/cmd_vel_legacy
/clicked_point
/clock
/goal_pose
/initialpose
/joy
/joy/set_feedback
/parameter_events
/pointCloud
/poses
/rosout
/tf
/tf_static
```

xiaoshan-lin, 2024-02-26:
Thank you for sharing these new updates. We are currently on a older commit, but will definitely check out the latest version!

Commenting `update_name_to_id_map(name, id); ` resolves our issue.

Thank you again for your help!

xiaoshan-lin, 2024-03-07:
To avoid sending the positions of non-crazyflie objects to the firmware, do we need to comment out any lines in  crazyflie_server.py, if we use the cflib backend?

closed 2024-03-20 by xiaoshan-lin