IMRCLab/crazyswarm2 #449: Orientation conflicts with Optitrack

MatCat960, 2024-03-08:
Hi all,

I'm facing some issues with physical tests using Optitrack. When I run `ros2 launch crazyflie launch.py` I get those warnings:

> motion_capture_tracking_node-1] [pcl::IterativeClosestPoint::computeTransformation] Not enough correspondences found. Relax your threshold parameters.
> [motion_capture_tracking_node-1] [WARN] [1709888165.079273882] [motion_capture_tracking]: ICP did not converge! for rigidBody cf4
> [motion_capture_tracking_node-1] [WARN] [1709888165.079329679] [motion_capture_tracking]: No updated pose for cf4 for 0.003371 s.

In RViz, I can see the position of the crazyflie is tracked correctly, but its orientation is continuously switching between two different configurations. Listening to the `\poses` topic returns a message containing sometimes one pose, sometimes two poses with different orientations: 

> poses:
> - name: cf4
>   pose:
>     position:
>       x: -0.08372469246387482
>       y: -0.460246741771698
>       z: 0.02166435308754444
>     orientation:
>       x: 0.03450774773955345
>       y: 0.034170813858509064
>       z: -0.021563591435551643
>       w: -0.9985873699188232
> ---
> header:
>   stamp:
>     sec: 0
>     nanosec: 0
>   frame_id: world
> poses:
> - name: cf4
>   pose:
>     position:
>       x: -0.08373195677995682
>       y: -0.4602496027946472
>       z: 0.021678747609257698
>     orientation:
>       x: 0.034297265112400055
>       y: 0.03398529812693596
>       z: -0.021510202437639236
>       w: -0.9986020922660828
> - name: cf4
>   pose:
>     position:
>       x: -0.08726711571216583
>       y: -0.4327550232410431
>       z: 0.018785975873470306
>     orientation:
>       x: 0.5671541690826416
>       y: -0.21594764292240143
>       z: -0.08667585253715515
>       w: 0.7900568842887878
> ---

I already tried to disable firmware logging from `crazyflie.yaml` config file but I'm still getting this strange behaviour.

Thank you in advance,

whoenig, 2024-03-08:
Just to clarify: You use 4 markers on the drone and there is no asset defined in Motive?

MatCat960, 2024-03-09:
I use 4 markers on the drone, saved as a rigid body named 'cf4' in Motive.

**Update:** I managed to get a stable orientation by disabling `motion capture` in `crazyflie.yaml` for the corresponding robot type. Now tracking is ok, I can run `hello_world` (which I could not before, as sudden rotations led to dynamic constraints violations), but the drone immediately crashes. Only one run ended up without any crashes.

However, I did not have the opportunity to make many tests, so I don't have any idea where the problem comes from. I will investigate more as soon as possible.

whoenig, 2024-03-09:
That makes sense - this "motion_capture" flag needs to be renamed. All it means is that custom-tracking (via librigidbodytracking) is enabled. If using the vendor software, nothing special needs to be done (apart from running motion-capture-tracking node).

Keep us posted about the other crashes (perhaps another issue/discussion, depending on what you find).

MatCat960, 2024-03-11:
Ok the problem seems to be solved now, crashes were due to some problems on our side.

I'm closing the issue. Thank you!

closed 2024-03-11 by MatCat960