IMRCLab/crazyswarm2 #463: Does needs any deck do take off ?

YHuniv, 2024-04-02:
Hi ,

I was trying to take of my crazyflie 2.1 without any deck.I did :

`sudo` apt-get install ros-humble-teleop-twist-keyboard

Termianl1 : 
`ros2` launch crazyflie_examples keyboard_velmux_launch.py
Terminal 2
`ros2` run teleop_twist_keyboard telop_twist_keyboard

Then, I used T to take off, but drone falls to the side. 
Did I missed something ?

Thanks by advance.

From : 
https://imrclab.github.io/crazyswarm2/tutorials.html
https://www.bitcraze.io/documentation/tutorials/getting-started-with-crazyflie-2-x/

knmcguire, 2024-04-02:
Based on your title 'Does it needs any deck do take off ?', yes it does if you want it to have positioning. Either there is an absolute positioning system: https://www.bitcraze.io/documentation/system/positioning/ or the flowdeck for relative positioning: https://www.bitcraze.io/products/flow-deck-v2/

YHuniv, 2024-04-02:
thanks for quick response.
That's mean the IMU estimator did not allow to take off even  estimator: 2 ? 

```
all:
  # firmware logging for all drones (use robot_types/type_name to set per type, or
  # robots/drone_name to set per drone)
  firmware_logging:
    enabled: true
    default_topics:
     # remove to disable default topic
      pose:
        frequency: 10 # Hz
      status:
        frequency: 1 # Hz
    #custom_topics:
    #  topic_name1:
    #    frequency: 10 # Hz
    #    vars: ["stateEstimateZ.x", "stateEstimateZ.y", "stateEstimateZ.z", "pm.vbat"]
    #  topic_name2:
    #    frequency: 1 # Hz
    #    vars: ["stabilizer.roll", "stabilizer.pitch", "stabilizer.yaw"]
  # firmware parameters for all drones (use robot_types/type_name to set per type, or
  # robots/drone_name to set per drone)
  firmware_params:
    commander:
      enHighLevel: 1
    stabilizer:
      estimator: 2 # 1: complementary, 2: kalman
      controller: 2 # 1: PID, 2: mellinger
```