IMRCLab/crazyswarm2 #494: colcon build error

shrutarv, 2024-05-06:
Hello,

I am getting the below error when doing a colcon build:
colcon build --symlink-install --cmake-args -DCMAKE_BUILD_TYPE=Release

Error:
This may be promoted to an error in a future release of colcon-override-check.
[1.286s] ERROR:colcon:colcon build: Duplicate package names not supported:
- CMSISDSP:
  - src/crazyswarm2/crazyflie-firmware/vendor/CMSIS/CMSIS/DSP/PythonWrapper
  - src/crazyswarm2/crazyflie-firmware/vendor/CMSIS/CMSIS/DSP/Source

I don't understand the error. It says duplicate packages but Source and PythonWrapper are different. Could someone please help me resolve it.

Thanks

knmcguire, 2024-05-06:
Hi! I haven't seen this before and the CI build seems to pass okay.

 Did you also put another clone of the crazyflie-firmware repo in the /src as well?

knmcguire, 2024-07-23:
closing this due to inactivity and not able to reproduce. Any support questions please ask them in discussions.

closed 2024-07-23 by knmcguire