IMRCLab/crazyswarm2 #518: _cffirmware error

GustavOliMoura, 2024-06-06:
### **I'm getting this error when trying to launch the launch.py on simulation with rviz2. Rviz opens but nothing shows on the visualization and i get those errors on the console:**

```
[INFO] [launch]: All log files can be found below /home/gustavo/.ros/log/2024-06-06-03-11-42-748399-gustavo-Aspire-VX5-22397
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [crazyflie_server-3]: process started with pid [22402]
[INFO] [teleop-1]: process started with pid [22398]
[INFO] [joy_node-2]: process started with pid [22400]
[INFO] [rviz2-4]: process started with pid [22404]
[teleop-1] [ERROR] [1717654303.011306888] [teleop]: Unknown mode none
[crazyflie_server-3] Traceback (most recent call last):
[crazyflie_server-3]   File "/home/gustavo/ros2_ws/install/crazyflie_sim/lib/crazyflie_sim/crazyflie_server", line 33, in <module>
[crazyflie_server-3]     sys.exit(load_entry_point('crazyflie-sim==2.0.0', 'console_scripts', 'crazyflie_server')())
[crazyflie_server-3]   File "/home/gustavo/ros2_ws/install/crazyflie_sim/lib/crazyflie_sim/crazyflie_server", line 25, in importlib_load_entry_point
[crazyflie_server-3]     return next(matches).load()
[crazyflie_server-3]   File "/usr/lib/python3.10/importlib/metadata/__init__.py", line 171, in load
[crazyflie_server-3]     module = import_module(match.group('module'))
[crazyflie_server-3]   File "/usr/lib/python3.10/importlib/__init__.py", line 126, in import_module
[crazyflie_server-3]     return _bootstrap._gcd_import(name[level:], package, level)
[crazyflie_server-3]   File "<frozen importlib._bootstrap>", line 1050, in _gcd_import
[crazyflie_server-3]   File "<frozen importlib._bootstrap>", line 1027, in _find_and_load
[crazyflie_server-3]   File "<frozen importlib._bootstrap>", line 1006, in _find_and_load_unlocked
[crazyflie_server-3]   File "<frozen importlib._bootstrap>", line 688, in _load_unlocked
[crazyflie_server-3]   File "<frozen importlib._bootstrap_external>", line 883, in exec_module
[crazyflie_server-3]   File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
[crazyflie_server-3]   File "/home/gustavo/ros2_ws/install/crazyflie_sim/local/lib/python3.10/dist-packages/crazyflie_sim/crazyflie_server.py", line 26, in <module>
[crazyflie_server-3]     from .crazyflie_sil import CrazyflieSIL, TrajectoryPolynomialPiece
[crazyflie_server-3]   File "/home/gustavo/ros2_ws/install/crazyflie_sim/local/lib/python3.10/dist-packages/crazyflie_sim/crazyflie_sil.py", line 10, in <module>
[crazyflie_server-3]     import cffirmware as firm
[crazyflie_server-3]   File "/home/gustavo/.local/lib/python3.10/site-packages/cffirmware.py", line 15, in <module>
[crazyflie_server-3]     import _cffirmware
[crazyflie_server-3] ModuleNotFoundError: No module named '_cffirmware'
[ERROR] [crazyflie_server-3]: process has died [pid 22402, exit code 1, cmd '/home/gustavo/ros2_ws/install/crazyflie_sim/lib/crazyflie_sim/crazyflie_server --ros-args -r __node:=crazyflie_server --params-file tmp_server.yaml'].
[rviz2-4] [INFO] [1717654303.599120779] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-4] [INFO] [1717654303.599330312] [rviz2]: OpenGl version: 4.6 (GLSL 4.6)
[rviz2-4] [INFO] [1717654303.624928001] [rviz2]: Stereo is NOT SUPPORTED
```

Eric-Ewing, 2024-06-06:
You need to install and set up the crazyflie firmware code for software-in-the-loop simulation. Step 7 here: https://imrclab.github.io/crazyswarm2/installation.html#first-installation

You may want to add the export PYTHONPATH command to your .bashrc file so it is set up correctly every time you open a terminal.

GustavOliMoura, 2024-06-06:
I have done it, actually it was working a few times i tried before showing this error. Do you think i might delete the crazyflie-firmware directory and install again?

Eric-Ewing, 2024-06-06:
You shouldn't need to reinstall it if it was previously working.

What does your PYTHONPATH look like? Does it contain the cffirmware directory?

GustavOliMoura, 2024-06-06:
Actually the pythonpath was wrong i changed the order from home to my user and it wasn't working but now it is thank you for observing that.

closed 2024-06-06 by GustavOliMoura