IMRCLab/crazyswarm2 #526: error when it comes to "sudo apt-get install ros-*DISTRO*-tf-transformations"

ugrm, 2024-06-20:
[image]

`jeahn@jeahn-VirtualBox:~$ pip3 install cflib transforms3d

Defaulting to user installation because normal site-packages is not writeable

Requirement already satisfied: cflib in ./.local/lib/python3.10/site-packages (0.1.25.1)

Requirement already satisfied: transforms3d in ./.local/lib/python3.10/site-packages (0.4.2)

Requirement already satisfied: libusb-package~=1.0 in ./.local/lib/python3.10/site-packages (from cflib) (1.0.26.2)

Requirement already satisfied: pyusb>=1.0.0b2 in ./.local/lib/python3.10/site-packages (from cflib) (1.2.1)

Requirement already satisfied: numpy~=1.20 in /usr/lib/python3/dist-packages (from cflib) (1.21.5)

Requirement already satisfied: scipy~=1.7 in /usr/lib/python3/dist-packages (from cflib) (1.8.0)

Requirement already satisfied: importlib-resources in ./.local/lib/python3.10/site-packages (from libusb-package~=1.0->cflib) (6.4.0)

jeahn@jeahn-VirtualBox:~$ sudo apt-get install ros-*DISTRO*-tf-transformations

[sudo] password for jeahn: 

Reading package lists... Done

Building dependency tree... Done

Reading state information... Done

E: Unable to locate package ros-*DISTRO*-tf-transformations

E: Couldn't find any package by glob 'ros-*DISTRO*-tf-transformations'

E: Couldn't find any package by regex 'ros-*DISTRO*-tf-transformations'

jeahn@jeahn-VirtualBox:~$ ^C

jeahn@jeahn-VirtualBox:~$ 

`

This is what the results are for running `pip3 install cflib transforms3d
sudo apt-get install ros-*DISTRO*-tf-transformations` . 
The reason I'm doing it on virtualbox is because I tried to make Ubuntu a dualboot system with windows, but I cannot get into safe mode.

ivankaDov, 2024-06-21:
I have the same problem. Working in Linux Ubuntu 22.04
`sudo apt-get ros-*DISTRO*-motion-capture-tracking
E: Invalid operation ros-*DISTRO*-motion-capture-tracking
`
I have not found the solving yet.

knmcguire, 2024-06-22:
Hi! 

*DISTRO* you'll need to replace with your ROS installation, like humble, iron etc.  

So for you it needs to be:

```
sudo apt-get install ros-humble-motion-capture-tracking
```

closed 2024-06-22 by whoenig

ugrm, 2024-06-23:
I think 
`sudo apt-get ros-DISTRO-motion-capture-tracking`
needs to be changed into 
`sudo apt-get **install** ros-DISTRO-motion-capture-tracking`
though. Just a heads up.

knmcguire, 2024-06-24:
Jup! you are right. I've added it to a latest PR for that https://github.com/IMRCLab/crazyswarm2/issues/526#event-13251592566

PR #527 (2024-06-22): Explain more clearly that distro needs to be replaced on install
Fixes #526