IMRCLab/crazyswarm2 #528: ROS2 Tutorials Teleoperation keyboard - package not found

ugrm, 2024-06-23:
- Ubuntu 22.04LTS
- ros humble
- running on virtualbox

I'm trying to do the teleoperation keyboard part of the ROS2 tutorials. 
When I run this line `ros2 launch crazyflie_examples keyboard_velmux_launch.py` inside ros2_ws/src I get the following error
`
jeahn@jeahn-VirtualBox:~/ros2_ws/src$ ros2 launch crazyflie_examples keyboard_velmux_launch.py
Package 'crazyflie_examples' not found: "package 'crazyflie_examples' not found, searching: ['/opt/ros/humble']"
jeahn@jeahn-VirtualBox:~/ros2_ws/src$ ^C
`
This is the case when I run some of the example python files like hello_world.py using `python3 hello_world.py`. 
I checked and I can find crazyflie_examples inside the ros2_ws file.

[image]

ugrm, 2024-06-23:
I reran colcon build, source /opt/ros/humble/setup.bash just in case and these didn't do anything

ugrm, 2024-06-23:
Running this code `ros2 launch crazyflie launch.py` in vizualization with RVIZ2 gives the same result
[image]

jeahn@jeahn-VirtualBox:~/ros2_ws/src$ ros2 launch crazyflie launch.py

Package 'crazyflie' not found: "package 'crazyflie' not found, searching: ['/opt/ros/humble']"

ugrm, 2024-06-24:
```
cd ros2_ws
cd src
colcon build
source install/setup.bash
```
Did this before running the ros2 launch code and it worked

closed 2024-06-24 by ugrm

knmcguire, 2024-06-24:
hi! yes if you build local packages you'll need to source it locally. It's not clear in the documentation so I'll add it

knmcguire, 2024-06-24:
ah no, it is already quite clear in https://imrclab.github.io/crazyswarm2/usage.html. So I'll leave it like this