IMRCLab/crazyswarm2 #535: Problems in a simple mapping with SLAMtoolbox

matheeusper, 2024-06-27:
After following the entire crazswarm2 tutorial and documentation, when doing a simple mapping with SLAMtoolbox, I have the following problem:
[image]
The same problem in #523 , i have already updated the repo and checked the file crazyflye.yaml, but I have the same problem:
[image]
There are no arguments odom and scam.
When run 
```
source install/setup.bash
ros2 launch crazyflie_examples multiranger_mapping_launch.py
``` 
Terminal output:
```
matheus@matheus-162082313018973:~$ cd ros2_ws/
matheus@matheus-162082313018973:~/ros2_ws$ 
matheus@matheus-162082313018973:~/ros2_ws$ source install/setup.bash
ros2 launch crazyflie_examples multiranger_mapping_launch.py
[INFO] [launch]: All log files can be found below /home/matheus/.ros/log/2024-06-26-23-08-55-772159-matheus-162082313018973-49230
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [crazyflie_server.py-1]: process started with pid [49231]
[INFO] [vel_mux.py-2]: process started with pid [49233]
[INFO] [async_slam_toolbox_node-3]: process started with pid [49235]
[async_slam_toolbox_node-3] [INFO] [1719454135.890291142] [slam_toolbox]: Using solver plugin solver_plugins::CeresSolver
[crazyflie_server.py-1] [INFO] [1719454138.099129607] [crazyflie_server]: [cf231] is fully connected!
[crazyflie_server.py-1] [INFO] [1719454138.099598474] [crazyflie_server]: All Crazyflies are fully connected!
[crazyflie_server.py-1] [INFO] [1719454138.105397396] [crazyflie_server]: [cf231] commander.enHighLevel is set to 1
[crazyflie_server.py-1] [INFO] [1719454138.135701315] [crazyflie_server]: [cf231] locSrv.extPosStdDev is set to 0.001
[crazyflie_server.py-1] [INFO] [1719454138.136544168] [crazyflie_server]: [cf231] locSrv.extQuatStdDev is set to 0.05
[crazyflie_server.py-1] [INFO] [1719454138.160506170] [crazyflie_server]: [cf231] stabilizer.controller is set to 1
[crazyflie_server.py-1] [INFO] [1719454138.161381732] [crazyflie_server]: [cf231] stabilizer.estimator is set to 2
[crazyflie_server.py-1] [INFO] [1719454138.169887842] [crazyflie_server]: All Crazyflies parameters are initialized.
[crazyflie_server.py-1] [INFO] [1719454138.171648098] [crazyflie_server]: [cf231] setup logging for pose at freq 10
[crazyflie_server.py-1] [INFO] [1719454138.172321192] [crazyflie_server]: [cf231] setup logging for status at freq 1
[crazyflie_server.py-1] [INFO] [1719454138.174304476] [crazyflie_server]: All Crazyflies logging are initialized.

```

knmcguire, 2024-06-27:
Hi! 

You'll need to add those variables to the firmware logging yourself, in the same way that pose and status are done. You could consider also just replacing pose and status with scan and odom and increase the last frequency to 10. 

Make sure that you also update the topic names in the multiranger_mapping_launch.py file, as it needs to point to the right crazyflie name and topics

Make sure to go through the usage documents of Crazyswarm2: https://imrclab.github.io/crazyswarm2/usage.html, as that explains how to add logging variables

matheeusper, 2024-06-27:
I needed to change the topic names in the multiranger_mapping_launch.py and it worked, thank you very much!