IMRCLab/crazyswarm2 #544: Proper usage to upload multiple trajectories

ywang760, 2024-07-12:
Hi, I have a question on using `cf.uploadTrajectory` to upload multiple trajectories. For example, I'm changing [this](https://github.com/IMRCLab/crazyswarm2/blob/main/crazyflie_examples/crazyflie_examples/figure8.py#L24) to 

```
for cf in allcfs.crazyflies:
            cf.uploadTrajectory(0, 0, traj)
            cf.uploadTrajectory(1, 0, traj1)
```
I'm getting
```
[crazyflie_server.py-4]   File "/home/yutongw/ros2_ws/install/crazyflie/lib/crazyflie/crazyflie_server.py", line 990, in _upload_trajectory_callback
[crazyflie_server.py-4]     trajectory_mem = self.swarm._cfs[uri].cf.mem.get_mems(
[crazyflie_server.py-4] IndexError: tuple index out of range
```
on the server.

knmcguire, 2024-07-23:
hmmm that is interesting, perhaps that is a bug on the cflib backend side. 

Does this work for the cpp backend? If you start the server with `backend:=cpp` ?

knmcguire, 2024-07-23:
There was a bug indeed, but luckily easy to spot the issue. There is a PR here https://github.com/IMRCLab/crazyswarm2/pull/549, if you can try it out?

ywang760, 2024-07-23:
Thank you. It works with `backend:=cpp`. Adding in https://github.com/IMRCLab/crazyswarm2/pull/549 also makes `backend:=cflib` works.

knmcguire, 2024-07-24:
Great! The fix has been merged so you can pull the latest changes on your machine.

PR #549 (2024-07-24): fix bug multiple trajectory upload
For some reason the mems retrieval was set to ID, but that is supposed to be the first index always.

This should fix #544
closed 2024-07-24 by knmcguire