IMRCLab/crazyswarm2 #545: Autonomous mapping

matheeusper, 2024-07-17:
How can I use the wall_following script with multiranger deck together with ROS to do automatic mapping?

knmcguire, 2024-07-20:
Hi! Sorry for the delay! The wall following node is temporary here: https://github.com/knmcguire/ros_gz_crazyflie. This is with simulation so you'll need to connect the topics yourself to those that are connected to the crazyswarm2 server. 

I recommend to try out the simple mapper tutorial of crazyswarm2 doc first before trying this

matheeusper, 2024-07-20:
[image]
When building crazyflie_sim package, there is the following error

matheeusper, 2024-07-20:
[image]
When trying to run the repo example I get the following error

knmcguire, 2024-07-23:
You'll need to source the install/setup.bash first. 

It would be best to first go to the ROS2 beginner tutorials first, which is necessary before trying out these examples: https://docs.ros.org/en/humble/Tutorials.html

Also, I will close this issue as this is not an project that is related to crazyswarm2. You should open up an issue at the repo I mentioned before instead.

closed 2024-07-23 by knmcguire