IMRCLab/crazyswarm2 #553: Empy 4.0.0 incompatible

knmcguire, 2024-08-09:
I've just did a fresh install of crazyswarm2 on my 24.04 so I'll make an issue of those that are still current.

When installing, usually empy needs to be installed as well with `pip3 install empy` which is currently missing from our documentation. This is necessary for building the crazyflie_interfaces (rosidl). 

However, there is currently an incompatibility with Empy 4.0.0 which causes these two errors:

```
...
    File "/opt/ros/jazzy/lib/python3.12/site-packages/rosidl_adapter/resource/__init__.py", line 51, in evaluate_template
      em.BUFFERED_OPT: True,
      ^^^^^^^^^^^^^^^

  AttributeError: module 'em' has no attribute 'BUFFERED_OPT'

...

    File "/opt/ros/jazzy/lib/python3.12/site-packages/rosidl_adapter/resource/__init__.py", line 69, in evaluate_template
      _interpreter.shutdown()
      ^^^^^^^^^^^^^^^^^^^^^

  AttributeError: 'NoneType' object has no attribute 'shutdown'
```

## solution

This is due to this error: https://github.com/ros2/rosidl/issues/779

Downgrade empy to:
```
pip3 install empy==3.3.4
```

suojinhui, 2024-11-18:
But I am using python 3.12, which is not compatible with empy==3.3.4

knmcguire, 2024-11-18:
aii... that's not awesome. And unfortunately no bug fix in sigh at the original issue at rosidl.

But what do you mean with incompatible? I could install this pip package for 3.12. What kind of error messages are you getting?

rdlynx19, 2025-02-04:
Hi @knmcguire
I'm trying to setup crazyswarm2 on my system running Ubuntu 24.04 and ROS2 Jazzy. I followed the instructions in the docs, but I'm getting a similar error when I run `colcon build`
Here are the logs:
```
(crazyflie-clients-python) redhairedlynx@tokyo:~/crazyflie_ws$ colcon build --cmake-args -DC_BUILD_TYPE=Release                                                        
Starting >>> crazyflie_interfaces                                                                                                                                      
--- stderr: crazyflie_interfaces                                                                                                                                       
CMake Error at /opt/ros/jazzy/share/rosidl_adapter/cmake/rosidl_adapt_interfaces.cmake:57 (message):                                                                   
                                                                                                                                                                       
  execute_process(/home/redhairedlynx/gitClones/crazyflie-clients-python/bin/python3                                                                                   
  -m rosidl_adapter --package-name crazyflie_interfaces --arguments-file                                                                                               
  /home/redhairedlynx/crazyflie_ws/build/crazyflie_interfaces/rosidl_adapter__arguments__crazyflie_interfaces.json                                                                                                                                                    
  --output-dir                                                                                                                                                         
  /home/redhairedlynx/crazyflie_ws/build/crazyflie_interfaces/rosidl_adapter/crazyflie_interfaces                                                                                                                                                                     
  --output-file                                                  
  /home/redhairedlynx/crazyflie_ws/build/crazyflie_interfaces/rosidl_adapter/crazyflie_interfaces.idls)                                                                                                                                                               
  returned error code 1:                                         
                                         
  AttributeError processing template 'msg.idl.em'                                                                                  
                                                                                   
  Traceback (most recent call last):                                               
                                                                                                                                                                       
    File "<frozen runpy>", line 198, in _run_module_as_main                                                                        
    File "<frozen runpy>", line 88, in _run_code                                                                                   
    File "/opt/ros/jazzy/lib/python3.12/site-packages/rosidl_adapter/__main__.py", line 19, in <module>
      sys.exit(main())              
               ^^^^^^               
    File "/opt/ros/jazzy/lib/python3.12/site-packages/rosidl_adapter/main.py", line 53, in main              
      abs_idl_file = convert_to_idl(
                     ^^^^^^^^^^^^^^^
    File "/opt/ros/jazzy/lib/python3.12/site-packages/rosidl_adapter/__init__.py", line 19, in convert_to_idl        
      return convert_msg_to_idl(                                                   
             ^^^^^^^^^^^^^^^^^^^                                                                                                                                       
    File "/opt/ros/jazzy/lib/python3.12/site-packages/rosidl_adapter/msg/__init__.py", line 39, in convert_msg_to_idl                                                                                                                                                 
      expand_template('msg.idl.em', data, output_file, encoding='iso-8859-1')
    File "/opt/ros/jazzy/lib/python3.12/site-packages/rosidl_adapter/resource/__init__.py", line 29, in expand_template  
      content = evaluate_template(template_name, data)                                                                             
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^                                                                             
    File "/opt/ros/jazzy/lib/python3.12/site-packages/rosidl_adapter/resource/__init__.py", line 65, in evaluate_template                                                                                                                                             
      _interpreter = em.Interpreter(                                               
                     ^^^^^^^^^^^^^^                              
                                                   
  AttributeError: module 'em' has no attribute 'Interpreter'                                                                                                                                                
                                                                                                      
[... cut ...]
Failed   <<< crazyflie_sim [0.53s, exited with code 1]
Aborted  <<< crazyflie_py [0.88s]                                                          
Aborted  <<< crazyflie [0.87s]

Summary: 0 packages finished [1.04s]
  1 package failed: crazyflie_sim
  2 packages aborted: crazyflie crazyflie_py
  2 packages had stderr output: crazyflie crazyflie_sim
  1 package not processed
(crazyflie-clients-python) redhairedlynx@tokyo:~/crazyflie_ws$ 

```
I am confused about what's going wrong here. Any help will be appreciated! Thanks

whoenig, 2025-02-04:
Could you share your `pip list | grep empy` output?

rdlynx19, 2025-02-05:
~/crazyflie_ws$ pip list | grep empy
empy                                 3.3.4

knmcguire, 2025-02-06:
Do you have perhaps different versions of Python on your machine? ROS might be using another one than of which the regular pip acts from.

rdlynx19, 2025-02-06:
I ran a bunch of commands to check this. 
```
redhairedlynx@tokyo:/usr/bin$ sudo find /usr/bin -type f -executable -iname 'python*'
/usr/bin/python3.12
/usr/bin/python-argcomplete-check-easy-install-script
/usr/bin/python3.12-coverage
/usr/bin/python3-coverage
```
 Only `python3.12` is installed on my system

knmcguire, 2025-02-09:
Maybe you have the 'em' package also installed.

If you do `pip list | grep em`, what do you see?

rdlynx19, 2025-02-09:
Looks like that was the issue. Thank you! @knmcguire 
```
(crazyflie-clients-python) redhairedlynx@tokyo:~/gitClones/crazyflie-clients-python(master)$ pip list | grep em
demo-nodes-py                        0.33.5
em                                   0.4.0
empy                                 3.3.4
logging-demo                         0.33.5
quality-of-service-demo-py           0.33.5

```

knmcguire, 2025-02-10:
Fwew luckily it was an easy fix still. Thanks for being so responsive

rdlynx19, 2025-02-19:
I think we can close the issue. I don't have the permissions to do this.

closed 2025-02-19 by whoenig