IMRCLab/crazyswarm2 #561: Add versioning to the yaml file

knmcguire, 2024-09-03:
We should add a version to the yaml files for better maintainability.

I guess this issue is more about how we'd like to do it, do we just add 'version:' on the top? and do we only do it for the crazyflie.yaml or for all the others as well?

whoenig, 2024-09-04:
Duplicate of https://github.com/IMRCLab/crazyswarm2/issues/222.

I think we should add it to all files, but it isn't that trivial because some of the files get merged. For example, both server.yaml and crazyflies.yaml get merged to become the parameters of the server. When adding the versioning, we need to be careful that we don't "shadow" the version of another file.

knmcguire, 2024-09-04:
ah woops! I'll close this. And yes we need to have some kind of handling on this. We can now add the check here in this function at least: https://github.com/IMRCLab/crazyswarm2/blob/8a22066c072bba4d90eef7632b68469e60cde18c/crazyflie/launch/launch.py#L11

closed 2024-09-04 by knmcguire