IMRCLab/crazyswarm2 #570: server - cflib - launch race condition

whoenig, 2024-09-23:
See https://github.com/IMRCLab/crazyswarm2/discussions/566#discussioncomment-10703248

whoenig, 2025-04-30:
Copying the relevant part here:

We tried `ros2 launch crazyflie_examples launch.py script:=hello_world backend:=cflib` with one crazyflie, but it's even worse: the crazyflie usually cannot successfully take off. It flies up and down, and sometimes even crashes into the ceiling. 
However, if we use two terminals(terminal1: `ros2 launch crazyflie launch.py backend:=cflib`, terminal2: `ros2 run crazyflie_examples hello_world`), cflib works well for one or two crazyflies.
Then why does `ros2 launch crazyflie_examples launch.py script:=hello_world backend:=cflib` not work even for one crazyflie? We didn't realize that it could be a problem because `ros2 launch crazyflie_examples launch.py script:=hello_world` seems well for one crazyflie.

knmcguire, 2025-05-05:
The problem I found, is that the service is not enabled yet because the crazyflie is not connected. It takes a good 3 seconds at least as a base start... and probably way more with more crazyflies. 

The cflib is just very slow with initial connection... so I'm not sure how to fix this except for adding a warning?

I'll dig in more into this

whoenig, 2025-05-05:
In the cpp backend, the "trick" is to enable/create the services only after the connection is done. The user scripts have a wait_for_service at the beginning to make sure that they only start running after the server is up and running.

PR #683 (2025-05-05): CFlib backend only create services after all crazyflies are fully connected
Currently in the server, the links are opened and then the services are already started, but it usually takes a couple of seconds for the crazyflies to be actually connected and react on those services. Since crazyflie_py class waits for services to be created, it will send any outgoing services too soon.

Fixes #570
closed 2025-05-05 by knmcguire