IMRCLab/crazyswarm2 #571: KeyError: 'firmware_params'

ddwy, 2024-09-26:
When I run the crazyswarm keyboard control command, the following error appears

```
bitcraze@bitcraze-vm:~/crazyflie_mapping_demo/ros2_ws$ ros2 launch crazyflie_examples keyboard_velmux_launch.py 
[INFO] [launch]: All log files can be found below /home/bitcraze/.ros/log/2024-09-26-09-47-52-527749-bitcraze-vm-26256
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [crazyflie_server.py-1]: process started with pid [26257]
[INFO] [vel_mux.py-2]: process started with pid [26259]
[crazyflie_server.py-1] Traceback (most recent call last):
[crazyflie_server.py-1]   File "/home/bitcraze/crazyflie_mapping_demo/ros2_ws/install/crazyflie/lib/crazyflie/crazyflie_server.py", line 1156, in <module>
[crazyflie_server.py-1]     main()
[crazyflie_server.py-1]   File "/home/bitcraze/crazyflie_mapping_demo/ros2_ws/install/crazyflie/lib/crazyflie/crazyflie_server.py", line 1147, in main
[crazyflie_server.py-1]     crazyflie_server = CrazyflieServer()
[crazyflie_server.py-1]   File "/home/bitcraze/crazyflie_mapping_demo/ros2_ws/install/crazyflie/lib/crazyflie/crazyflie_server.py", line 141, in __init__
[crazyflie_server.py-1]     self.swarm.query_all_values_on_connect = self._ros_parameters["firmware_params"]["query_all_values_on_connect"]
[crazyflie_server.py-1] KeyError: 'firmware_params'
[ERROR] [crazyflie_server.py-1]: process has died [pid 26257, exit code 1, cmd '/home/bitcraze/crazyflie_mapping_demo/ros2_ws/install/crazyflie/lib/crazyflie/crazyflie_server.py --ros-args -r __node:=crazyflie_server --params-file /tmp/launch_params_arbe_qoe'].
```

ddwy, 2024-09-27:
> Could you share the content of your crazyflie.yaml file? it seems there is an error in there.

The following is the content of the crazyflie.yaml file:

```
# named list of all robots
robots:
  cf231:
    enabled: true
    uri: radio://0/80/2M/E7E7E7E7E7
    initial_position: [0.0, 0.0, 0.0]
    type: cf21  # see robot_types
    # firmware_params:
    #   kalman:
    #     pNAcc_xy: 1.0 # default 0.5
    # firmware_logging:
    #   enabled: true
    #   custom_topics:
    #     topic_name3: 
    #       frequency: 1
    #       vars: ["acc.x", "acc.y"]
  cf5:
    enabled: false
    uri: radio://0/80/2M/E7E7E7E705
    initial_position: [0.0, -0.5, 0.0]
    type: cf21  # see robot_types
    # firmware_params:
    #   kalman:
    #     pNAcc_xy: 1.0 # default 0.5
    #firmware_logging:
    #  custom_topics:
    #    topic_name3: 
    #      frequency: 1
    #      vars: ["acc.x", "acc.y", "acc.z"]
# Definition of the various robot types
robot_types:
  cf21:
    motion_capture:
      enabled: true
      # only if enabled; see motion_capture.yaml
      marker: default_single_marker
      dynamics: default
    big_quad: false
    battery:
      voltage_warning: 3.8  # V
      voltage_critical: 3.7 # V
    # firmware_params:
    #   kalman:
    #     pNAcc_xy: 1.0 # default 0.5
    #firmware_logging:
    #   enabled: true
    #   default_topics:
    #   pose:
    #     frequency: 1 # Hz
    #   custom_topics:
    #    topic_name3: 
    #      frequency: 1
    #      vars: ["acc.x", "acc.y", "acc.z"]

  cf21_mocap_deck:
    motion_capture:
      enabled: true
      # only if enabled; see motion_capture.yaml
      marker: mocap_deck
      dynamics: default
    big_quad: false
    battery:
      voltage_warning: 3.8  # V
      voltage_critical: 3.7 # V

    # firmware_params:
    #   kalman:
    #     pNAcc_xy: 1.0 # default 0.5

# global settings for all robots
all:
  # firmware logging for all drones (use robot_types/type_name to set per type, or
  # robots/drone_name to set per drone)
  firmware_logging:
    enabled: true
    default_topics:
     # remove to disable default topic
      pose:
        frequency: 10 # Hz
      status:
        frequency: 1 # Hz
    #custom_topics:
    #  topic_name1:
    #    frequency: 10 # Hz
    #    vars: ["stateEstimateZ.x", "stateEstimateZ.y", "stateEstimateZ.z", "pm.vbat"]
    #  topic_name2:
    #    frequency: 1 # Hz
    #    vars: ["stabilizer.roll", "stabilizer.pitch", "stabilizer.yaw"]
  # firmware parameters for all drones (use robot_types/type_name to set per type, or
  # robots/drone_name to set per drone)
  firmware_params:
    commander:
      enHighLevel: 1
    stabilizer:
      estimator: 2 # 1: complementary, 2: kalman
      controller: 2 # 1: PID, 2: mellinger
    # ring:
    #   effect: 16 # 6: double spinner, 7: solid color, 16: packetRate
    #   solidBlue: 255 # if set to solid color
    #   solidGreen: 0 # if set to solid color
    #   solidRed: 0 # if set to solid color
    #   headlightEnable: 0
    locSrv:
      extPosStdDev: 1e-3
      extQuatStdDev: 0.5e-1
    # kalman:
    #   resetEstimation: 1
  broadcasts:
    num_repeats: 15 # number of times broadcast commands are repeated
    delay_between_repeats_ms: 1 # delay in milliseconds between individual repeats

```

knmcguire, 2024-09-27:
Ah yes that is a bug! I'll look into it right away and notify you if I have a PR out.

knmcguire, 2024-09-27:
Could you try out the branch of this PR https://github.com/IMRCLab/crazyswarm2/pull/574 @ddwy ?

PR #574 (2024-10-01): Fix parameter seperation issues cflib server
All the crazyflie examples use the include launch description now such that I don't have to recreate the parameter handling everytime when something is changed.

 Should fix #571
closed 2024-10-01 by knmcguire