IMRCLab/crazyswarm2 #577: [gui.py-5] Error

hitmyself, 2024-10-13:
When I run the following command,：

`ros2 launch crazyflie launch.py`

It gets:

```
[gui.py-5] Error:   TF_NAN_INPUT: Ignoring transform for child_frame_id "cf4" from authority "default_authority" because of a nan value in the transform (-0.007230 -0.012239 0.009829) (nan nan nan nan)
[gui.py-5]          at line 235 in ./src/buffer_core.cpp
[gui.py-5] Error:   TF_DENORMALIZED_QUATERNION: Ignoring transform for child_frame_id "cf4" from authority "default_authority" because of an invalid quaternion in the transform (nan nan nan nan)
[gui.py-5]          at line 254 in ./src/buffer_core.cpp``
```
how can i solve this problem

whoenig, 2024-10-16:
Is the functionality affected? This seems to be some warning (even though it says error), that is most likely happening in single marker tracking mode (where quaternions are set to nan).

knmcguire, 2025-04-29:
closed due to inactivity