IMRCLab/crazyswarm2 #591: building ROS2 space

ntukenmez3, 2024-11-17:
We would like to set-up experimental environment for swarm operation and then we started to follow the Crazyswarm2 site.
We have gotten the following error. Could you help us to fix it? Thanks.
[image]

ntukenmez3, 2024-11-17:
The answer is that ROS2 is supposed to be sourced.

`source /opt/ros/humble/setup.bash`

ntukenmez3, 2024-11-17:
> We would like to set-up experimental environment for swarm operation and then we started to follow the Crazyswarm2 site. We have gotten the following error. Could you help us to fix it? Thanks. [...]

closed 2024-11-17 by whoenig

knmcguire, 2024-11-18:
This is related to this issue as well. I'll make a PR of the documentation ASAP