IMRCLab/crazyswarm2 #592: building ROS 2 workspace

ntukenmez3, 2024-11-17:
We'd like to build our ROS2 workspace with motion capture system.
We installed the mocap package separately as given below.
However, it seems something is wrong with this workspace. Could you help with this one? Thanks.

```
ntukenmez3@aelicps08120l:~/Desktop/ros2_ws/src$ git clone --recurse-submodules https://github.com/IMRCLab/motion_capture_tracking
Cloning into 'motion_capture_tracking'...
remote: Enumerating objects: 390, done.
remote: Counting objects: 100% (80/80), done.
remote: Compressing objects: 100% (24/24), done.
remote: Total 390 (delta 64), reused 60 (delta 56), pack-reused 310 (from 1)
Receiving objects: 100% (390/390), 63.92 KiB | 3.36 MiB/s, done.
Resolving deltas: 100% (202/202), done.
Submodule 'deps/libmotioncapture' (https://github.com/IMRCLab/libmotioncapture.git) registered for path 'motion_capture_tracking/deps/libmotioncapture'
Submodule 'deps/librigidbodytracker' (https://github.com/IMRCLab/librigidbodytracker.git) registered for path 'motion_capture_tracking/deps/librigidbodytracker'
Cloning into '/home/ntukenmez3/Desktop/ros2_ws/src/motion_capture_tracking/motion_capture_tracking/deps/libmotioncapture'...
remote: Enumerating objects: 898, done.        
remote: Counting objects: 100% (227/227), done.        
remote: Compressing objects: 100% (53/53), done.        
remote: Total 898 (delta 191), reused 174 (delta 174), pack-reused 671 (from 1)        
Receiving objects: 100% (898/898), 148.14 KiB | 3.70 MiB/s, done.
Resolving deltas: 100% (495/495), done.
Cloning into '/home/ntukenmez3/Desktop/ros2_ws/src/motion_capture_tracking/motion_capture_tracking/deps/librigidbodytracker'...
remote: Enumerating objects: 895, done.        
remote: Counting objects: 100% (207/207), done.        
remote: Compressing objects: 100% (55/55), done.        
remote: Total 895 (delta 165), reused 175 (delta 148), pack-reused 688 (from 1)        
Receiving objects: 100% (895/895), 379.53 KiB | 6.12 MiB/s, done.
Resolving deltas: 100% (512/512), done.
Submodule path 'motion_capture_tracking/deps/libmotioncapture': checked out '53d759c6cecf217a12bcd664cf4264c4127400a1'
Submodule 'externalDependencies/NatNetSDKCrossplatform' (https://github.com/whoenig/NatNetSDKCrossplatform.git) registered for path 'motion_capture_tracking/deps/libmotioncapture/deps/NatNetSDKCrossplatform'
Submodule 'externalDependencies/pybind11' (https://github.com/pybind/pybind11.git) registered for path 'motion_capture_tracking/deps/libmotioncapture/deps/pybind11'
Submodule 'externalDependencies/qualisys_cpp_sdk' (https://github.com/qualisys/qualisys_cpp_sdk.git) registered for path 'motion_capture_tracking/deps/libmotioncapture/deps/qualisys_cpp_sdk'
Submodule 'externalDependencies/vicon-datastream-sdk' (https://github.com/whoenig/vicon-datastream-sdk.git) registered for path 'motion_capture_tracking/deps/libmotioncapture/deps/vicon-datastream-sdk'
Submodule 'externalDependencies/vrpn' (https://github.com/vrpn/vrpn.git) registered for path 'motion_capture_tracking/deps/libmotioncapture/deps/vrpn'
Cloning into '/home/ntukenmez3/Desktop/ros2_ws/src/motion_capture_tracking/motion_capture_tracking/deps/libmotioncapture/deps/NatNetSDKCrossplatform'...
remote: Enumerating objects: 133, done.        
remote: Counting objects: 100% (121/121), done.        
remote: Compressing objects: 100% (76/76), done.        
remote: Total 133 (delta 46), reused 108 (delta 34), pack-reused 12 (from 1)        
Receiving objects: 100% (133/133), 1.31 MiB | 11.99 MiB/s, done.
Resolving deltas: 100% (48/48), done.
Cloning into '/home/ntukenmez3/Desktop/ros2_ws/src/motion_capture_tracking/motion_capture_tracking/deps/libmotioncapture/deps/pybind11'...
remote: Enumerating objects: 30008, done.        
remote: Counting objects: 100% (1372/1372), done.        
remote: Compressing objects: 100% (631/631), done.        
remote: Total 30008 (delta 883), reused 1025 (delta 647), pack-reused 28636 (from 1)        
Receiving objects: 100% (30008/30008), 12.01 MiB | 29.00 MiB/s, done.
Resolving deltas: 100% (21135/21135), done.
Cloning into '/home/ntukenmez3/Desktop/ros2_ws/src/motion_capture_tracking/motion_capture_tracking/deps/libmotioncapture/deps/qualisys_cpp_sdk'...
remote: Enumerating objects: 110, done.        
remote: Counting objects: 100% (110/110), done.        
remote: Compressing objects: 100% (70/70), done.        
remote: Total 110 (delta 66), reused 63 (delta 40), pack-reused 0 (from 0)        
Receiving objects: 100% (110/110), 130.77 KiB | 3.53 MiB/s, done.
Resolving deltas: 100% (66/66), done.
Cloning into '/home/ntukenmez3/Desktop/ros2_ws/src/motion_capture_tracking/motion_capture_tracking/deps/libmotioncapture/deps/vicon-datastream-sdk'...
remote: Enumerating objects: 370, done.        
remote: Counting objects: 100% (212/212), done.        
remote: Compressing objects: 100% (121/121), done.        
remote: Total 370 (delta 125), reused 154 (delta 86), pack-reused 158 (from 1)        
Receiving objects: 100% (370/370), 445.87 KiB | 7.31 MiB/s, done.
Resolving deltas: 100% (213/213), done.
Cloning into '/home/ntukenmez3/Desktop/ros2_ws/src/motion_capture_tracking/motion_capture_tracking/deps/libmotioncapture/deps/vrpn'...
remote: Enumerating objects: 25079, done.        
remote: Counting objects: 100% (610/610), done.        
remote: Compressing objects: 100% (298/298), done.        
remote: Total 25079 (delta 335), reused 536 (delta 297), pack-reused 24469 (from 1)        
Receiving objects: 100% (25079/25079), 23.33 MiB | 29.60 MiB/s, done.
Resolving deltas: 100% (14926/14926), done.
[... 482 lines cut ...]
                 from /usr/include/boost/asio/basic_deadline_timer.hpp:25,
                 from /usr/include/boost/asio.hpp:25,
                 from /home/ntukenmez3/Desktop/ros2_ws/src/motion_capture_tracking/motion_capture_tracking/deps/libmotioncapture/include/libmotioncapture/fzmotion.h:7,
                 from /home/ntukenmez3/Desktop/ros2_ws/src/motion_capture_tracking/motion_capture_tracking/deps/libmotioncapture/src/motioncapture.cpp:25:
/usr/include/boost/bind.hpp:36:1: note: ‘#pragma message: The practice of declaring the Bind placeholders (_1, _2, ...) in the global namespace is deprecated. Please use <boost/bind/bind.hpp> + using namespace boost::placeholders, or define BOOST_BIND_GLOBAL_PLACEHOLDERS to retain the current behavior.’
   36 | BOOST_PRAGMA_MESSAGE(
      | ^~~~~~~~~~~~~~~~~~~~
In file included from /home/ntukenmez3/Desktop/ros2_ws/src/motion_capture_tracking/motion_capture_tracking/deps/libmotioncapture/src/motioncapture.cpp:25:
/home/ntukenmez3/Desktop/ros2_ws/src/motion_capture_tracking/motion_capture_tracking/deps/libmotioncapture/include/libmotioncapture/fzmotion.h:133:32: warning: ISO C++ prohibits anonymous structs [-Wpedantic]
  133 |                         struct { uint8 v1; uint8 v2; uint8 v3; uint8 v4; };
      |                                ^
/home/ntukenmez3/Desktop/ros2_ws/src/motion_capture_tracking/motion_capture_tracking/deps/libmotioncapture/include/libmotioncapture/fzmotion.h:138:32: warning: ISO C++ prohibits anonymous structs [-Wpedantic]
  138 |                         struct { uint8 v1; uint8 v2; uint8 v3; uint8 v4; };
      |                                ^
/home/ntukenmez3/Desktop/ros2_ws/src/motion_capture_tracking/motion_capture_tracking/deps/libmotioncapture/include/libmotioncapture/fzmotion.h:145:32: warning: ISO C++ prohibits anonymous structs [-Wpedantic]
  145 |                         struct { octet h1; octet h2; octet l1; octet l2; };
      |                                ^
---
Finished <<< motion_capture_tracking [35.6s]
Finished <<< crazyflie [43.7s]                       

Summary: 7 packages finished [47.0s]
  4 packages had stderr output: crazyflie_examples crazyflie_py crazyflie_sim motion_capture_tracking

```

ntukenmez3, 2024-11-17:
When we updated the local copy, we got the following output:

```
ntukenmez3@aelicps08120l:~/Desktop/ros2_ws$ colcon build --symlink-install --cmake-args -DCMAKE_BUILD_TYPE=Release
Starting >>> crazyflie_interfaces
Starting >>> motion_capture_tracking_interfaces
Finished <<< motion_capture_tracking_interfaces [0.49s]                                                                          
Starting >>> motion_capture_tracking
Finished <<< motion_capture_tracking [0.31s]                                                                      
Finished <<< crazyflie_interfaces [0.86s]
Starting >>> crazyflie_py
Starting >>> crazyflie
Starting >>> crazyflie_sim
Finished <<< crazyflie [0.41s]                                                                                    
--- stderr: crazyflie_sim                                                             
/usr/lib/python3/dist-packages/setuptools/command/install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
---
Finished <<< crazyflie_sim [0.60s]
--- stderr: crazyflie_py                   
/usr/lib/python3/dist-packages/setuptools/command/easy_install.py:158: EasyInstallDeprecationWarning: easy_install command is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
---
Finished <<< crazyflie_py [0.77s]
Starting >>> crazyflie_examples
--- stderr: crazyflie_examples                         
/usr/lib/python3/dist-packages/setuptools/command/install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
---
Finished <<< crazyflie_examples [0.34s]

Summary: 7 packages finished [2.16s]
  3 packages had stderr output: crazyflie_examples crazyflie_py crazyflie_sim

```

whoenig, 2024-11-17:
These are warnings, not errors. Some of them can (and will) be fixed, other warnings are generated by third-party libraries.

closed 2024-11-17 by whoenig