IMRCLab/crazyswarm2 #598: cmdVel(self, roll, pitch, yawrate, thrust) publisher

ntukenmez3, 2024-11-25:
We'd like to use `cmdVel` publisher for our experiment scenario. However, it looks this publisher has been prepared for ROS with `rospy`.
For ROS:

`self.cmdVelPublisher = rospy.Publisher(prefix + '/cmd_vel', geometry_msgs.msg.Twist, queue_size=1)`

I guess we should make it with `rclpy` for ROS2.
So, to do that, we will need `crazyflie_interfaces.msg` for `geometry_msgs.msg`. I put question marks where we need help.
For ROS2:

```
self.cmdVelPublisher = node.create_publisher(???, prefix + '/cmd_vel', 1)
self.??? = ???
self.???.header.frame_id = '/world'
```

Could you help us with this one? 
And also are we gonna need to configure these files :` crazyflie_interfaces.msg` and `crazyflie_interfaces.srv` ?

whoenig, 2024-11-25:
The crazyflie server has not been prepared for cmd_vel, so logic would need to be added there first (Python and C++). The Python layer is fairly simple and can follow existing logic, see e.g., https://github.com/IMRCLab/crazyswarm2/blob/5e731c91d25bcc883205340801ac379280423abb/crazyflie_py/crazyflie_py/crazyflie.py#L578-L621.

ntukenmez3, 2024-11-25:
Yes, I think so. I've already started on it.
The point I need some help with is:
`self.cmdFullStateMsg = FullState()`
for `cmd_vel`, which function should I use in place of `FullState` since this function comes from from `crazyflie_interfaces.msg import FullState, Position, Status, TrajectoryPolynomialPiece` ?

ntukenmez3, 2024-11-25:
I guess I can use still `FullState()` then I can pick which I need among the functions.
However, I couldn't catch one part, I added below:
```
msg = geometry_msgs.msg.Twist()
msg.linear.x = pitch
msg.linear.y = roll
msg.angular.z = yawrate
msg.linear.z = thrust
self.cmdVelPublisher.publish(msg)
```

How can we exactly see the relation between `pitch` input and `x` displacement or `roll` input and `y` displacement?
Could you also explain this part if it's possible?

ntukenmez3, 2024-11-26:
Should the topic name be `/cmd_vel_legacy` since `crazyflie_server.py` shows the following topic name?

```
self.create_subscription(
                Twist, name +
                "/cmd_vel_legacy", partial(self._cmd_vel_legacy_changed,
                                           uri=uri), 10
            )
```
And, should we add `Twist.msg` in `crazyflie_interfaces/msg/` ?

ntukenmez3, 2024-11-27:
I've prepared two different publishers, however none of them has worked. I've added below:
The first one:
```
self.cmdVelPublisher = node.create_publisher(FullState, prefix + '/cmd_vel_legacy', 1)
        self.cmdVelMsg = FullState()
        self.cmdVelMsg.header.frame_id = '/world'
```
its function:
```
def cmdVel(self, roll, pitch, yawrate, thrust):
        
        self.cmdVelMsg.header.stamp = self.node.get_clock().now().to_msg()
        self.cmdVelMsg.twist.linear.x = pitch
        self.cmdVelMsg.twist.linear.y = roll
        self.cmdVelMsg.twist.linear.z = thrust
        self.cmdVelMsg.twist.angular.z = yawrate
        self.cmdVelPublisher.publish(self.cmdVelMsg)

```

The second one:

```
self.cmdVelPublisher = node.create_publisher(Twist, prefix + '/cmd_vel_legacy', 1)
        self.cmdVelMsg = Twist()
        self.cmdVelMsg.header.frame_id = '/world'
```

its function:
```
def cmdVel(self, roll, pitch, yawrate, thrust):
        
        self.cmdVelMsg.header.stamp = self.node.get_clock().now().to_msg()
        self.cmdVelMsg.linear.x = pitch
        self.cmdVelMsg.linear.y = roll
        self.cmdVelMsg.linear.z = thrust
        self.cmdVelMsg.angular.z = yawrate
        self.cmdVelPublisher.publish(self.cmdVelMsg)

```

Could you help us to fix it?

whoenig, 2024-11-27:
You need to first send 0 commands to "unlock", and then send your desired commands in a loop at a reasonable rate (I'd say at least 10 Hz). Please also check with `ros2 topic echo` if you are indeed publishing on the topic. Moreover, you have to use the cflib backend (since the default cpp backend has not implemented that mode).

knmcguire, 2024-11-28:
Alright, it is for us a bit unclear what you'd like to achieve, because what cmd_vel_legacy is, is attitude control with roll/pitch/raw/thrust with Twist messages. However, that is not what is standard within ROS, who use cmd_vel with Twist velocity messages. But that is what you want right?

Have you looked at the twist keyboard examples in the documentation? https://imrclab.github.io/crazyswarm2/tutorials.html#teleoperation-keyboard

It emulates a planar velocity movement from a /cmd_vel topic, by utilizing the cmd_hover topic with with a simple multiplexer script. Please take a look at the launch file here: https://github.com/IMRCLab/crazyswarm2/blob/main/crazyflie_examples/launch/keyboard_velmux_launch.py

ntukenmez3, 2024-11-28:
We would like to send `(roll, pitch, yaw rate thrust)` command (contol inputs) to crazyflies in the swarm network (regarding our experiment scenario) through ROS2, that is exactly what we want.

ntukenmez3, 2024-11-29:
> The crazyflie server has not been prepared for cmd_vel, so logic would need to be added there first (Python and C++). The Python layer is fairly simple and can follow existing logic, see e.g., [...]

This is exactly what we need to do to use `/cmd_vel_legacy`, thank you @whoenig.

knmcguire, 2024-12-02:
We will close this then. 

Just for next time, could you consider not removing your posts in this issue? Our answers to some of your removed questions or comments don't really make sense now, plus we also got a quite some notifications of the comments that were removed.