IMRCLab/crazyswarm2 #618: Crazyswarm2 SlamToolboxPlugin ros2 humble can't save map files .data, .posegraph, .pgm, .yaml

wladimirpetrov, 2025-01-10:
**Required Info:**

- Operating System:
  - Ubuntu 22.04.5 LTS
- Installation type:
  - ROS 2 packages and CrazySwarm2 installed from source
- ROS Version
  - ROS 2 Humble
- Version or commit hash:
  - Using the latest release of Slam Toolbox and CrazySwarm2 from their respective repositories.
  
#### Steps to reproduce issue
1. Start the Crazyflie server using the configured crazyflies.yaml file and the multiranger_nav2_launch.py launch file.
2. Open RViz2 and add the SLAM Toolbox plugin.
3. Attempt to save the map using the "Save Map" button in the SLAM Toolbox plugin interface.
4. Alternatively, try saving the map via the CLI:
`ros2 service call /slam_toolbox/save_map slam_toolbox/srv/SaveMap "{name: '~/ros2_ws/maps/map'}"
`
5. Another attempt was made using:
`ros2 run nav2_map_server map_saver_cli -f ~/ros2_ws/maps/map --ros-args -r map:=/cf231/map -p map_subscribe_transient_local:=true
`

#### Expected behavior
When saving the map using either the SLAM Toolbox plugin in RViz2 or via the CLI, the following files should be generated:
.pgm
.yaml
.data
.posegraph
These files are required for localization and further map-related tasks.

#### Actual behavior
1. Clicking the "Save Map" button in the SLAM Toolbox plugin does nothing. The terminal shows the following warning:
`[WARN] [SlamToolboxPlugin]: SlamToolbox: Failed to save map as , is service running?
`
2. Attempting to call the /slam_toolbox/save_map service results in the following error:
`Failed to populate field: Value '~/ros2_ws/maps/map' is expected to be a dictionary but is a str
`
3. Using the nav2_map_server map_saver_cli command successfully generates the .pgm and .yaml files, but the .data and .posegraph files are not generated.

#### Additional information
Here is the list of nodes running during the process:
`/SlamToolboxPlugin
/SlamToolboxStateUpdateNode
/camera/camera
/crazyflie_server
/rviz
/teleop_twist_keyboard
/transform_listener_impl_613584c3b3b0
/vel_mux
`
List of active topics:
`/cf231/cmd_full_state
/cf231/cmd_hover
/cf231/cmd_vel_legacy
/cf231/map
/cf231/map_updates
/cf231/odom
/cf231/pose
/cf231/robot_description
/cf231/scan
/cf231/status
/cmd_vel
/initialpose
/parameter_events
/poses
/rosout
/tf
/tf_static
`
Conclusion
While the SLAM Toolbox is able to generate .pgm and .yaml files using the nav2_map_server map_saver_cli, the .data and .posegraph files are not being created. Using the SLAM Toolbox plugin in RViz2 also fails to save the map entirely, with the warning message indicating the service might not be running

wladimirpetrov, 2025-01-10:
in addition to the previous message.
After I saved the map in two files usigng command:
`ros2 run nav2_map_server map_saver_cli -f ~/ros2_ws/maps/map --ros-args -r map:=/cf231/map -p map_subscribe_transient_local:=true `

I try to check wheether i can open it using Nav2, and i get the following:
`petrov@intern97dt:~/ros2_ws$ ros2 launch crazyflie_examples multiranger_nav2_launch.py
[INFO] [launch]: All log files can be found below /homes/petrov/.ros/log/2025-01-10-09-35-11-931901-intern97dt-95036
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [crazyflie_server.py-1]: process started with pid [95050]
[INFO] [vel_mux.py-2]: process started with pid [95052]
[INFO] [async_slam_toolbox_node-3]: process started with pid [95054]
[INFO] [component_container_isolated-4]: process started with pid [95056]
[INFO] [rviz2-5]: process started with pid [95058]
[rviz2-5] Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
[async_slam_toolbox_node-3] [INFO] [1736519712.185542751] [slam_toolbox]: Using solver plugin solver_plugins::CeresSolver
[async_slam_toolbox_node-3] [ERROR] [1736519712.189050987] [slam_toolbox]: serialization::Read: Failed to open requested file: /homes/petrov/ros2_ws/maps/map.yaml.
[async_slam_toolbox_node-3] [ERROR] [1736519712.189079878] [slam_toolbox]: DeserializePoseGraph: Failed to read file: /homes/petrov/ros2_ws/maps/map.yaml.
`
However when open the saved map file it seem to look exactly like it should: 
`image: map.pgm
mode: trinary
resolution: 0.1
origin: [-10, -10, 0]
negate: 0
occupied_thresh: 0.65
free_thresh: 0.25`

knmcguire, 2025-01-13:
Hi! Thanks for the report and the map saving has not been tested for a long time. We won't have time this week to check this out, but we will update here. If you managed to find the solution, please let us know.

wladimirpetrov, 2025-01-13:
Thank you for your help! Sure, I'll let you know in case the problem is solved

wladimirpetrov, 2025-01-14:
So I kept on playing with the error I had, decided to re-install the slam toolbox for ros2 humble. After a couple of iterations it started to work. 
So there is no any bug, sorry for bothering you!

wladimirpetrov, 2025-01-15:
problem was solved

closed 2025-01-15 by wladimirpetrov

knmcguire, 2025-01-16:
thanks for letting us know!