IMRCLab/crazyswarm2 #658: 2025.02 firmware cflib kalman reset not working

wladimirpetrov, 2025-03-26:
Hello. 

I am using 2025.02 crazyflie-firmware with crazyswarm2 cflib. In my code I do:
`cf.setParam("kalman.resetEstimation", 1)`

and it throws an error:

```
param_type = self.paramTypeDict[name]
KeyError: 'kalman.resetEstimation'
```
when i try to print:

`print(cf.paramTypeDict.keys())`

I get:

`dict.keys([])`

What am I doing in the wrong way? According to the documentation it should work.

Thank you for your time!

knmcguire, 2025-03-30:
Hi! Yes that is indeed strange and it should still be there according to the autodoc: https://www.bitcraze.io/documentation/repository/crazyflie-firmware/master/api/params/#kalmanresetestimation

Could you put this parameter in the server.yaml on True and try again?

https://github.com/IMRCLab/crazyswarm2/blob/f0f914657e14efbc6580229d171d4cdf76a3385e/crazyflie/config/server.yaml#L14

knmcguire, 2025-04-01:
Thanks! Just mind that now connecting will take a bit longer with the Crazyflie, but with a single one that should be fine.

closed 2025-04-29 by knmcguire