IMRCLab/crazyswarm2 #752: Trajectory command not respond , maybe broadcast problem.

duguguang, 2025-09-24:
When using Crazyswarm2 (commit 70f3f1060e75139a5f1d380b8a809bb89b63ccc5) in conjunction with a motion capture system, even with just one drone, the trajectory function fails to respond properly. Comparative tests reveal that cflib works as expected when serving as the backend, but Crazyswarm2 experiences significant communication issues. 

Environment configuration 
Crazyswarm2 version: Main branch latest (70f3f1060e75139a5f1d380b8a809bb89b63ccc5) 
Crazyflie firmware: 202502 
Operating System: Ubuntu 20.04 
ROS2: Galactic

Hardware: AMD Ryzen 5 7600x, 32GB RAM 
Crazyradio 2.0 firmware: Multiple versions have been tested. 
1.1 release version (original) 
A custom version for fixing the issue of repeated broadcasts 
The version with enhanced interrupt priority 
Based on the Zephyr 4.2 version

Reproduction steps: 

1. ros2 launch crazyflie launch.py
2. python hello_world.py
3. python figure8_csv.py 

Result:
hello_world  is ok but figure8 is not respond.

On fix arc version(based on tag 1.1), https://github.com/bitcraze/crazyradio2-firmware/pull/11:
`ros2 launch crazyflie launch.py 
[INFO] [launch]: All log files can be found below /home/nokov/.ros/log/2025-09-24-17-03-31-398084-nokov-288745
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [motion_capture_tracking_node-1]: process started with pid [288755]
[INFO] [crazyflie_server-2]: process started with pid [288757]
[crazyflie_server-2] [INFO] [1758704611.654251662] [crazyflie_server]: [cf50] Requesting parameters...
[crazyflie_server-2] [INFO] [1758704611.661802932] [crazyflie_server]: [cf50] Param TOC: 310 entries with CRC 3679191350
[crazyflie_server-2] [INFO] [1758704611.661846028] [crazyflie_server]: [cf50] Param TOC: found cache.
[motion_capture_tracking_node-1] logClouds=0
[crazyflie_server-2] [INFO] [1758704611.671737167] [crazyflie_server]: [cf50] reqParamTOC: 0.020847 s (310 params)
[crazyflie_server-2] [INFO] [1758704611.671830699] [crazyflie_server]: [cf50] Update parameter "commander.enHighLevel" to 1
[crazyflie_server-2] [INFO] [1758704611.671873617] [crazyflie_server]: [cf50] Update parameter "locSrv.extPosStdDev" to 0.001000
[crazyflie_server-2] [INFO] [1758704611.671911932] [crazyflie_server]: [cf50] Update parameter "locSrv.extQuatStdDev" to 0.050000
[crazyflie_server-2] [INFO] [1758704611.671947352] [crazyflie_server]: [cf50] Update parameter "stabilizer.controller" to 2
[crazyflie_server-2] [INFO] [1758704611.671981962] [crazyflie_server]: [cf50] Update parameter "stabilizer.estimator" to 2
[crazyflie_server-2] [INFO] [1758704611.672012987] [crazyflie_server]: [cf50] ref frame: world
[crazyflie_server-2] [INFO] [1758704611.681520081] [crazyflie_server]: [cf50] Log TOC: 525 entries with CRC 3880012195
[crazyflie_server-2] [INFO] [1758704611.681588003] [crazyflie_server]: [cf50] Log TOC: found cache.
[crazyflie_server-2] [INFO] [1758704611.682017012] [crazyflie_server]: [cf50] Logging to /pose at 10 Hz
[crazyflie_server-2] [INFO] [1758704611.689464739] [crazyflie_server]: [cf50] Logging to /status at 1 Hz
[crazyflie_server-2] [INFO] [1758704611.697363549] [crazyflie_server]: [cf50] Requesting memories...
[crazyflie_server-2] [INFO] [1758704611.701325668] [crazyflie_server]: [cf50] Memories: 4
[crazyflie_server-2] [INFO] [1758704611.768957785] [crazyflie_server]: [cf50] ESTKALMAN: State out of bounds, resetting
[crazyflie_server-2] [WARN] [1758704612.645606133] [crazyflie_server]: [all] Motion capture rate off (#: 21, Avg: 130.9, Min: 17.9, Max: 560.6)
[crazyflie_server-2] [WARN] [1758704612.650961263] [crazyflie_server]: [cf50] last latency update: 1.001549 s
[crazyflie_server-2] [WARN] [1758704612.697022434] [crazyflie_server]: [cf50] Unexpected number of unicast packets. Sent: 1353. Received: 54816
[crazyflie_server-2] [WARN] [1758704612.697045445] [crazyflie_server]: [cf50] Unexpected number of broadcast packets. Sent: 89. Received: 8602
[crazyflie_server-2] [WARN] [1758704613.645618543] [crazyflie_server]: [all] Motion capture rate off (#: 29, Avg: 145.2, Min: 12.6, Max: 600.0)
[crazyflie_server-2] [WARN] [1758704613.699985297] [crazyflie_server]: [cf50] Low unicast receive rate (0.87 < 0.90). Sent: 1271. Received: 1108
[crazyflie_server-2] [WARN] [1758704613.700011414] [crazyflie_server]: [cf50] Unexpected number of broadcast packets. Sent: 85. Received: 244
[crazyflie_server-2] [WARN] [1758704614.645635464] [crazyflie_server]: [all] Motion capture rate off (#: 26, Avg: 131.2, Min: 11.2, Max: 467.5)
[crazyflie_server-2] [WARN] [1758704614.651048754] [crazyflie_server]: [cf50] last latency update: 1.996072 s
[crazyflie_server-2] [WARN] [1758704614.695002524] [crazyflie_server]: [cf50] Unexpected number of broadcast packets. Sent: 68. Received: 158
[crazyflie_server-2] [WARN] [1758704615.645663582] [crazyflie_server]: [all] Motion capture rate off (#: 2, Avg: 90.7, Min: 72.2, Max: 115.9)
[crazyflie_server-2] [WARN] [1758704615.694062700] [crazyflie_server]: [cf50] Unexpected number of broadcast packets. Sent: 14. Received: 43
[crazyflie_server-2] [WARN] [1758704617.645688069] [crazyflie_server]: [all] Motion capture rate off (#: 10, Avg: 103.5, Min: 18.8, Max: 471.8)
[crazyflie_server-2] [WARN] [1758704617.694094724] [crazyflie_server]: [cf50] Low unicast receive rate (0.85 < 0.90). Sent: 1156. Received: 983
[crazyflie_server-2] [WARN] [1758704617.694128128] [crazyflie_server]: [cf50] Unexpected number of broadcast packets. Sent: 87. Received: 232
[crazyflie_server-2] [WARN] [1758704618.645711564] [crazyflie_server]: [all] Motion capture rate off (#: 12, Avg: 106.4, Min: 22.9, Max: 446.0)
[crazyflie_server-2] [WARN] [1758704618.651093161] [crazyflie_server]: [cf50] last latency update: 1.996067 s
[crazyflie_server-2] [WARN] [1758704618.692069278] [crazyflie_server]: [cf50] Low unicast receive rate (0.86 < 0.90). Sent: 1170. Received: 1005
[crazyflie_server-2] [WARN] [1758704618.692096638] [crazyflie_server]: [cf50] Unexpected number of broadcast packets. Sent: 90. Received: 248
[crazyflie_server-2] [WARN] [1758704619.645774359] [crazyflie_server]: [all] Motion capture rate off (#: 11, Avg: 98.1, Min: 14.8, Max: 459.9)
[crazyflie_server-2] [WARN] [1758704619.651083769] [crazyflie_server]: [cf50] last latency update: 2.996060 s
[crazyflie_server-2] [WARN] [1758704619.691126062] [crazyflie_server]: [cf50] Low unicast receive rate (0.86 < 0.90). Sent: 1240. Received: 1068
[... 286 lines cut ...]
[crazyflie_server-2] [WARN] [1758704719.620106963] [crazyflie_server]: [cf50] Low unicast receive rate (0.88 < 0.90). Sent: 1215. Received: 1068
[crazyflie_server-2] [WARN] [1758704719.620138731] [crazyflie_server]: [cf50] Unexpected number of broadcast packets. Sent: 90. Received: 261
[crazyflie_server-2] [WARN] [1758704719.647482499] [crazyflie_server]: [all] Motion capture rate off (#: 28, Avg: 100.9, Min: 31.4, Max: 330.4)
[crazyflie_server-2] [WARN] [1758704720.618534110] [crazyflie_server]: [cf50] Low unicast receive rate (0.89 < 0.90). Sent: 1293. Received: 1145
[crazyflie_server-2] [WARN] [1758704720.618564579] [crazyflie_server]: [cf50] Unexpected number of broadcast packets. Sent: 89. Received: 227
[crazyflie_server-2] [WARN] [1758704720.647558137] [crazyflie_server]: [all] Motion capture rate off (#: 23, Avg: 104.9, Min: 41.8, Max: 404.4)
[crazyflie_server-2] [WARN] [1758704721.618085609] [crazyflie_server]: [cf50] Low unicast receive rate (0.89 < 0.90). Sent: 1187. Received: 1058
[crazyflie_server-2] [WARN] [1758704721.618119879] [crazyflie_server]: [cf50] Unexpected number of broadcast packets. Sent: 90. Received: 292
[crazyflie_server-2] [WARN] [1758704721.647522442] [crazyflie_server]: [all] Motion capture rate off (#: 22, Avg: 100.8, Min: 36.8, Max: 392.9)
[crazyflie_server-2] [WARN] [1758704722.618955238] [crazyflie_server]: [cf50] Low unicast receive rate (0.87 < 0.90). Sent: 1283. Received: 1110
[crazyflie_server-2] [WARN] [1758704722.618986244] [crazyflie_server]: [cf50] Unexpected number of broadcast packets. Sent: 84. Received: 215
[crazyflie_server-2] [WARN] [1758704722.647561793] [crazyflie_server]: [all] Motion capture rate off (#: 24, Avg: 99.5, Min: 33.5, Max: 385.4)
[crazyflie_server-2] [WARN] [1758704723.618070878] [crazyflie_server]: [cf50] Low unicast receive rate (0.89 < 0.90). Sent: 1293. Received: 1154
[crazyflie_server-2] [WARN] [1758704723.618098979] [crazyflie_server]: [cf50] Unexpected number of broadcast packets. Sent: 76. Received: 220
[... cut ...]
[crazyflie_server-2] [WARN] [1758704725.647584240] [crazyflie_server]: [all] Motion capture rate off (#: 16, Avg: 88.8, Min: 30.0, Max: 346.3)
[crazyflie_server-2] [WARN] [1758704726.617574248] [crazyflie_server]: [cf50] Low unicast receive rate (0.87 < 0.90). Sent: 1146. Received: 993
[crazyflie_server-2] [WARN] [1758704726.617603559] [crazyflie_server]: [cf50] Unexpected number of broadcast packets. Sent: 84. Received: 228
[crazyflie_server-2] [WARN] [1758704726.647630181] [crazyflie_server]: [all] Motion capture rate off (#: 24, Avg: 94.4, Min: 44.1, Max: 350.8)
`

knmcguire, 2025-09-24:
Hi! 

Mind that we no longer support ROS2 galactic. Anyway that you could try it for humble or higher? 

And do you have a [Crazyradio PA ](https://store.bitcraze.io/collections/accessories/products/crazyradio-pa)or a [Crazyradio 2.0](https://store.bitcraze.io/collections/accessories/products/crazyradio-2-0) ?

@whoenig, which chipsets did you see issues with with the Crazyradio 2? Where those AMD or Intel?

whoenig, 2025-09-24:
We have issues with Crazyradio2, independent of Chipset. For Crazyradio PA, it seems to work better on Intel computers.

duguguang, 2025-09-26:
> Hi!
>
> Mind that we no longer support ROS2 galactic. Anyway that you could try it for humble or higher?
>
> And do you have a [Crazyradio PA [...]

The issue with PA has been bothering us all along. We have tested the higher versions but found no improvement. Version 1.0 of PA works fine, but we only have one at hand, which is not enough for our daily research and testing. Recently, we were forced to switch to the backend method of CFLib, and ensure the crazyflie firmware syncd with cflib version(the same date publised by bitcraze) but new problems emerged when reading the setting parameters, such as the inability to effectively set groups.